upb-lea/LCB-CCB-01_LEA_Control_Board #11: U902 Input Voltage range

gituser789, 2024-10-02:
Common mode input voltage range is not enough.

[image]

gituser789, 2024-12-13:
* [ ] check needed current from analog part of the controller
* [ ] check provided current from LDO
* [ ] Remove the buffer

VijayThera, 2024-12-19:
**Removed buffer U902** 
[image]

needed current for analog part of the controller - Max. 20 mA
[image]

provided current from LDO - Typ. ~ 30 mA
[image]